marcel-licence/esp32_basic_synth #18: Multiplexer connection.

Opekusha, 2021-05-18:
Please tell me which Pin of ESP32 the output from the multiplexer is connected to.

marcel-licence, 2021-05-18:
```
#define ADC_MUL_S0_PIN	23 // pin 23 for S0 
#if ADC_INPUTS > 2
#define ADC_MUL_S1_PIN	18 // pin 18 for S1 
#endif
#if ADC_INPUTS > 4
#define ADC_MUL_S2_PIN	14 // pin 14 for S2
#endif
#if ADC_INPUTS > 8
#define ADC_MUL_S3_PIN	5	 /* this has not been tested */
#endif
```

`    adcAttachPin(12); // pin 12 is used for ADC input`

S3 goes to ground so you can use C0..C7 of the multiplexer.
EN should be also connected to ground.
SIG to pin 12 of the Audio Kit V2.2

Hope this helps

Opekusha, 2021-05-18:
I am using the first option with a PCM external DAC. I do not have an audio KIT. You need a pin for the ESP32 G *. The rest of the pins are understandable, but thanks anyway for your attention.

marcel-licence, 2021-05-18:
I've bring back the possibility to compile for the ESP32 DOIT DEVKIT. I think that will help. I've updated the readme. Please tell me if something missing

tiretrak, 2021-05-20:
which multiplexer to use?

tiretrak, 2021-05-20:
4051 is ok?