marcel-licence/esp32_basic_synth #19: Compilation problems

Slider2732, 2021-05-18:
Equipment setup is: ESP-WROOM32, MAX98357A I2S DAC, same 16 channel multiplexer as shown in the ADC enabled video. 
Issue summary: The code compiles as downloaded, but doesn't if the ESP32_AUDIO_KIT is deselected. Also problems if the AC101 isn't selected. Example - 'l_sample was not declared in this scope' in the I2S_NODAC section (meaning another DAC than the AC101?)

The original version of the code did compile with my parts and worked, had sound from MIDI keys. It only had my own lack of a MIDI controller as a problem. Delightfully fixed and demo'd in your video of the new ADC enabled version.
The audio kit seems to be uncommonly rare, otherwise would have been ordered.  
Rather than shooting in the dark, I'd like to know what actually needs to be changed... in the right way for perhaps a more average set of components   :)

marcel-licence, 2021-05-18:
You attempt is correct to deselect the ESP32_AUDIO_KIT. The actual version of the code is actually only supporting the audio kit. I think some little changes will be required to make it compatible with different boards.
Regarding the ADC .. please check the serial output. When pots some output should be generated.

Slider2732, 2021-05-18:
Thanks for the quick reply. Found the audio kit so have ordered one... arrives mid June. 
Will check the serial for ADC output in any case.
If you do get the chance, it would be a bonus to run these projects of yours on common hardware. I do understand that the audio kit is preferred though.

marcel-licence, 2021-05-18:
I've updated the code, now it should compile for the ESP32 DEVKIT again. I didn't have the setup to test it with the ADC. But it should work with the normal MIDI setup. I've also updated the readme. Please let me know if something is missing.

Slider2732, 2021-05-19:
WOW! Instant update and yes, it works. Thanks Marcel.
There's sound output, something like a sine wave  :)
#define ADC_TO_MIDI_ENABLED was then kept, but it said several variables weren't recognised. By adding the following, it compiled, though I haven't been able to verify my multiplexer wiring yet: 

#define analogSetCycles
#define adcStart
#define adcBusy
#define adcEnd

marcel-licence, 2021-05-19:
Perfect =)

It's only confusion that the linker/compiler couldn't find the adc functions. Maybe you have another version of the libraries than me.

Opekusha, 2021-05-19:
This is what the compiler gave me. I just did this: // # define ESP32_AUDIO_KIT to use PCM 5102.   'sampleDataU' was not declared in this scope . Where and what should you turn on or off? What libraries to download? In the first versions, everything was compiled. I'm lost...

marcel-licence, 2021-05-19:
Did you try to compile the latest code? I made an update yesterday.

Opekusha, 2021-05-19:
I checked everything for 10 out of 10 points. It remains to finish everything in jelly. Thank you so much. I look forward to your new works.

Opekusha, 2021-05-19:
The translation is not accurate. Sorry.

closed 2021-05-20 by marcel-licence