marcel-licence/esp32_basic_synth #27: ADC not seen, no changes in audio or serial output

Slider2732, 2021-05-20:
As hinted at in a previous issue (and thank you for that prompt attention), the ADC pots are unfortunately not recognised. Output through the DAC is only a non changing, but nice and mellow organ sound.  Enabling the debug output doesn't produce any changes in the serial output, just the initial values are shown. 
Equipment is: ESP32 Dev, 4067 16 channel Multiplexer, Max 98357A I2S DAC

The following lines were needed in the ADC section for the esp32_basic_synth to compile -
#define analogSetCycles
#define adcStart
#define adcEnd
#define adcBusy

The multiplexer does work, on the synth circuit, when tested with the code here:
https://gtsdtech.wordpress.com/2018/01/15/esp32-working-with-74hc4067-16-channel-mux-demux/
I changed the pins to be the ones in the esp32_basic_ synth and all was fine, LED's flashed as channels were stepped through. 

Have tried various changes, but decided to drop them and ask what may be happening. 
Thanks for any insight  :)

tiretrak, 2021-05-21:
try this schematic
[image]

tiretrak, 2021-05-21:
[image]

Slider2732, 2021-05-21:
A very useful diagram for others too, thanks firetrak.
That's how my own is wired up, and 8 ADC_INPUTS is correct. 
After checking all soldering, a test was to disconnect all 8 pots and try just one, with its own GND/Multiplexer/Vcc connections... no change. 

Am wondering now if the need to add the includes in the adc section is where I've messed up, though the synth code wouldn't compile without them.
Will try another same type multiplexer, just in case something happened to the other.

tiretrak, 2021-05-21:
try connecting one potentiometer C0 or C1 or C2. when cutoff is at minimum, the sound is very quiet

marcel-licence, 2021-05-21:
Thanks a lot for the picture. I did also some testing on the DOIT ESP32 DEVKIT. I got it now working. Actually the code works only with the older library: esp32\1.0.4.
I hope I find some time to prepare an update for 1.0.6 soon. I commited my changes but changed also the PIN's because I am preparing something new and had some conflicts otherwise.

The PIN settings are now also in the config.h. I will try to put all personal settings into the config file and the rest in the other files to make it easier to work with.

The cutoff at minimum will cut off all frequencies above. So it is normal that you the sound will disappear.

marcel-licence, 2021-05-21:
For just trying the ADC without multiplexer you could also set ADC_INPUTS to 1.
In addition to that set also ADC_TO_MIDI_LOOKUP_SIZE to 1.
 
And finally in z_config.ino let only one value inside of the lookup table:

struct adc_to_midi_s adcToMidiLookUp[ADC_TO_MIDI_LOOKUP_SIZE] =
{
    {0, 0x10},
};

Slider2732, 2021-05-21:
Ah OK, good clarification. 

I downgraded to 1.04 and now have some functionality!
The code compiles fine without the #includes.
C4 connected pot on the multiplexer is doing 'things', there is a large sonic change in output when it's rotated. Also, it displays the data changes in the serial output.

The next part will be getting the other 7 pots to work similarly. But, moving from 1.06 to 1.04 was indeed the fix.

Great troubleshooting, thanks guys  :)

closed 2021-05-23 by Slider2732