marcel-licence/esp32_basic_synth #32: Have added an octave up switch + code

Slider2732, 2021-05-24:
Hopefully of use to others and OK by Marcel to offer up as a change. 
I've added some simple lines to enable an octave shift of a keyboard by one octave upward.

The situation, was that I've built the synth to pair with a 3 octave 1970's PAiA 2720 keyboard bed. The unison mode was problematic, in that the notes were often too low for where I wanted to play them, it ran out of top end. The mod allows the keyboard to behave as though it has 4 octaves.
It could probably be done another way, but it works  :)
Pin is 13 on an ESP32 Dev (can be any pin). 10K to Ground, switch to 3.3V Vcc.

Changed code is in midi_interface.ino, plus 'pinMode (13, INPUT);' in void setup() :

inline void Midi_NoteOn(uint8_t ch, uint8_t note, uint8_t vel)
{
    if (digitalRead(13) == HIGH)
       {
          note = note +12; // raise an octave
       }

    if (vel > 127
    {
        /* we will end up here in case of problems with the MIDI connection */
        vel = 127
        Serial.printf("loud note detected!!!!!!!!!!!!!!!!!!!!!!!\n");
    }

    if (midiMapping.noteOn != NULL)
    { 
       if (digitalRead(13) == HIGH)
       midiMapping.noteOn(ch, note+12, pow(2, ((vel * NORM127MUL) - 1.0f) * 6));
    }  
    
    {
       if (digitalRead(13) == LOW)
       midiMapping.noteOn(ch, note, pow(2, ((vel * NORM127MUL) - 1.0f) * 6));
    }
}

inline void Midi_NoteOff(uint8_t ch, uint8_t note)
{
  if (digitalRead(13) == HIGH)
  {note = note +12; } // raise an octave
  
  if (midiMapping.noteOff != NULL)
  {
    {
       if (digitalRead(13) == HIGH)
       midiMapping.noteOff(ch, note+12);
    }  
      
    {
       if (digitalRead(13) == LOW)
       midiMapping.noteOff(ch, note);
    }
   }
}

marcel-licence, 2021-05-25:
Good point. It could be also a general transpose value in case you have less keys. Very intersting keyboard you are using.
I think it would be also possible to use the internal pull ups (I think some GPIO's should support it) then you can switch directly to ground. 
A pitch bend over multiple octave's would be also possible or an octave knob for the synth module itself. Just some ideas.
What do you think about using #define MIDI_OCTAVE_UP_PIN 13 in config.h? That would avoid some suprises inside the midi module itself.
Do you want to go above midi note 127 (13289.75Hz) or ist just your value in modo note on event to low?
The generator should match this table: https://www.inspiredacoustics.com/en/MIDI_note_numbers_and_center_frequencies

Thanks a lot for sharing

Slider2732, 2021-05-27:
Ah yes, an internal pull-up enabled pin would be better. 

4x PAiA keybeds were part of a lot that I bought locally. 3 have been MIDI enabled now, using Arduino's and diodes with a matrix. Many 70's synths seem to have used the same 3 octave keybed... which gave thoughts of people repurposing damaged or broken synths with your projects  :)
An octave knob would be a cool feature, selecting different input pins. I think people would prefer that over ranges on a pot, for the clicky feel factor.
A full on Mini Moog or Pro 1 emulation would be a project!

I'd be honoured if you incorporated the octave code into the synth engine!

13289.75Hz is no problem. I'm 51, I can hardly hear that high lol   
 It was just higher trance type lines that the keyboard ran out of notes for, such as part of Binary Finary.

marcel-licence, 2021-05-28:
Maybe you could also add a transpose functionality to your keybed reading arduinos? There are so many possibilities :-D But yes, adding some map-able parameters could be a good idea. So as optional features which can be mapped if wanted.

marcel-licence, 2021-11-21:
Now available with midi_ctrl.ino =)

closed 2021-11-21 by marcel-licence