marcel-licence/esp32_basic_synth #33: Velocity is not connected in the code. It uses a default 0.25

ErichHeinemann, 2021-05-26:
Hi Marcel, 
I dropped my code and replaced it by YOur new version as You added Pitchbend and Modwheel.
I missed the velocity and thought that my keyboards was not sending it. It was not cennected in the code.

in easySynth.ino

 inline void Synth_NoteOn(uint8_t ch, uint8_t note, float vel)
{
    struct notePlayerT *voice = getFreeVoice();
    struct oscillatorT *osc = getFreeOsc();

    /*
     * No free voice found, return otherwise crash xD
     */
    if ((voice == NULL) || (osc == NULL))
    {
        //Serial.printf("voc: %d, osc: %d\n", voc_act, osc_act);
        return ;
    }

    voice->midiNote = note;
    voice->velocity = vel * 1.25 ;// 0.25; /* just something to test */
    voice->lastSample[0] = 0.0f;

marcel-licence, 2021-05-26:
Hi Erich,

thanks, I've updated the code. By default MIDI_USE_CONST_VELOCITY is set which forces using const velocity (there are often cases where you do not want dynamic loudness of your notes).

But the curve in the MIDI module is not good. It feels a bit weird. I found for example this document with some information about the peak/peak to velocity curve: https://www.cs.cmu.edu/~rbd/papers/velocity-icmc2006.pdf

ErichHeinemann, 2021-05-26:
Hi Marcel,
as I played around with Your newer Synth-Code today, I miss a bit more Volume for lower notes and a bit less for higher notes.
From the Human Frequency-Curve, perhaps You could add this into the synth. … Notenumber to control the Volume … -( Notenumer / 10 ) or something else… 
Perhaps You could compare it with internal Sound of Your Roland Synth.
I cannot measure it but it is a strange synthetic feeling to hut the higher Keys less than the lower Keys.

perhaps we could find that in the code from other developers / projects?

Regards
Erich

marcel-licence, 2021-05-28:
I could try to measure. I just set the peak to peak value to a constant value. I am actually just testing with a 2.1 sound system where I can set the subwoofer level. I could imagine that a little curve/linear function could help to tweak. vel = a * note + b

closed 2021-06-03 by marcel-licence