marcel-licence/esp32_basic_synth #43: How to flash files with arduino ide

asttljsj, 2021-06-08:
Hello, how can I use arduino ide to burn files, and it prompts that delay-init is not defined during verification, thanks

marcel-licence, 2021-06-08:
Hello,
google might help: [https://www.google.com/search?client=firefox-b-d&q=esp32+arduino+how+to]
Regarding your error message, please paste in a copy. At the moment I do not have any idea what went wrong.
Best regards,

asttljsj, 2021-06-09:
[esp32_basic_synth.zip](https://github.com/marcel-licence/esp32_basic_synth/files/6620487/esp32_basic_synth.zip)
Hello, when I open it with arduino ide, it prompts that delay-init is not defined. What library files do you need for your project?

HilmarSt, 2021-06-22:
Delay_Init() is implemented in the file simple_delay.ino , no additional libraries are needed.
Just make sure that all .ino files from the repo are in the same directory as the esp32_basic_synth.ino file.

marcel-licence, 2021-06-22:
@HilmarSt thank you for posting the solution

closed 2021-06-22 by marcel-licence