marcel-licence/esp32_basic_synth #44: I2S_NODAC configuration: writeDAC() is not implemented

HilmarSt, 2021-06-22:
In the actual repo the writeDAC() function only exists as a prototype without the function itself (and of course will not compile).
Do you just miss to include it or is it not yet implemented?
I expected something like a pulse density modulated I2S, I've seen it before on https://www.hackster.io/janost/audio-hacking-on-the-esp8266-fa9464#toc-a-simple-909-drum-synth-0

marcel-licence, 2021-06-22:
The function in your link should work. I've also tested this function and I've found another version.
I think I've an implemented version maybo in another project. I tried to simplify it. I can check if I can update this project to have only one function call and a central config to choose using NO_DAC

HilmarSt, 2021-06-22:
I found the function also in your code, it's i2s_write_stereo_samples() in the #ifdef I2S_NODAC branch.

With the following modifications it compiles:

1. remove
#ifdef I2S_NODAC
    if (writeDAC(l_sample))
    {
        l_sample = Synth_Process();
    }
#else (and #endif)

2. remove adc_module.ino

3. copy
struct adc_to_midi_s
{
    uint8_t ch;
    uint8_t cc;
};
from adc_module.ino to z_config.ino

closed 2021-06-22 by HilmarSt

HilmarSt, 2021-06-22:
> I found the function also in your code, it's i2s_write_stereo_samples() in the #ifdef I2S_NODAC branch.
>
> With the following modifications it compiles:
>
>     1. remove
>        #ifdef I2S_NODAC [...]

My configuration is:
//#define ESP32_AUDIO_KIT
#define I2S_NODAC
//#define ADC_TO_MIDI_ENABLED