marcel-licence/esp32_basic_synth #45: I just have no clue

MichaelPNolan, 2021-07-05:
MIDI can be received via USB activating the MACRO "MIDI_VIA_USB_ENABLED" in config.h.
this is not in the sketch ... do i have to transplant the files from another module into the project?

marcel-licence, 2021-07-05:
This is a preparation for the usb midi integration. Actually there is a standalone project: https://github.com/marcel-licence/esp32_usb_midi
I think I will merge it soon into this project to support MIDI via usb

MichaelPNolan, 2021-07-05:
Ok - i got it to work - i integrated ESP32_usb_midi into ESP32_basic_synth
It was kind of not working until i put this into the midi_interface file of this project ..

`void Midi_SendShortMessage(uint8_t *msg)
{
    Serial2.write(msg, 3);
}`

You also have to be careful that you aren't using pins used for SPI because there are overlaps. I know you know, just saying for the purposes of this. Assuming you know because you have made all the things work together.
At some point my launchpad started to show a turn on sequence when I rebooted the ESP32.
Then I got printed out notes coming from the pad ... 
I made quite a few tweaks before this actually worked so I'm a bit confused what was the magic sauce but i have this in the loop.

 Midi_Process();
        //#ifdef MIDI_VIA_USB_ENABLED
             UsbMidi_ProcessSync();
        //#endif
        UsbMidi_Loop();

Do I need UsbMidi_ProcessSync(); there at the top level loop?

marcel-licence, 2021-07-06:
Sounds great. UsbMidi_ProcessSync is required to process incoming midi events in sync. I put them into a buffer and UsbMidi_ProcessSync gets them from the buffer. Otherwise I had a lot of trouble with midi interaction

MichaelPNolan, 2021-07-09:
So a DAC arrived ... i've been going more into it with discrete components rather than the Esp32-A1S-AudioDev board. I have a pcm5102 breakout. I have USB midi calling notes and sound playing. The sound seems like its a bit of a mess - lowest notes making noise higher notes very quiet ... So I guess its time to get to mapping things and see if I can get that to work. Also to solder together the analogue inputs for your ADC module of controls.

marcel-licence, 2021-07-20:
Code has been updated in the meantime

closed 2021-07-20 by marcel-licence