marcel-licence/esp32_basic_synth #57: Using Core0

MichaelPNolan, 2021-07-18:
I moved some processing to the Core0 task loop

At first it wasn't being called but I checked your code and it was updated to the MIDI_VIA_USB... module
so adding that in now it works.
```
#if (defined ADC_TO_MIDI_ENABLED) || (defined MIDI_VIA_USB_ENABLED)
    xTaskCreatePinnedToCore(Core0Task, "Core0Task", 8000, NULL, 999, &Core0TaskHnd, 0);
#endif
```

In the first 30-60 seconds I get audio glitches - even one where it was making a rough popping on left side then right side alternating.
So processing Core0 seems to slow auto processing which we assume only happens on Core1 in arduino.
Therefore, generally no improvement - but over time things seem to buffer or perform ok.
So I'll look at having a kind of reduced frequency of calls in the Task0loop  using some kind of loop count variable to see if that removes that glitch.

marcel-licence, 2021-07-18:
Do you run your own version actually or the latest, original code from GitHub?

MichaelPNolan, 2021-07-18:
I guess it's my own version at this point. So I'd have to map all my
changes to upgrade. Not that many changes, trying to use per design. Mostly
add ons.

Regards
Michael

MichaelPNolan, 2021-07-19:
> Hm okay when there are modifications then it would be very hard to
> support. I would like to convert this issue into a discussion then.

Ok - you made me realize that. However, i was just surprised since I was
doing it all without Core0Task and then I got ... less performance by using
it?
So that part ... is more like an issue but its not shared software. Was
just thinking you might have tips or have experienced same along the way.
Please convert.
Regards
Michael

closed 2021-07-20 by marcel-licence