marcel-licence/esp32_basic_synth #74: Can't compile Synthselect was not declared

KeenVox, 2021-11-12:
Hi, 
I can't get the code to compile. It always stops at "Syntselect was not declared in this scope". 
I'm using the Audikit V2.2.
Board in the manager ist ESP32 DEV Modul. Version 1.0.6.n 1.04 makes no difference. I tried some different arduino versions, but didn't work.
Any advice?
Thanks!

marcel-licence, 2021-11-12:
Hi,
thank you for this information. Looks like I was asleep when commiting the code.
I've fixed the build for the non fake organ version. I try to bring in the required code soon to run also the fake organ version.

marcel-licence, 2021-11-12:
Oh I was sleeping:
`void Synth_SetFader(uint8_t slider, float value)
{
    if ((slider < 0) && (slider < 9))`
...
that is crazy, that would never work

KeenVox, 2021-11-12:
Now it compiles. Thx!
Still I get no sound and midi doesn't seem to work. I'm using a Audio Kit V2.2 and have connected midi over an 6N138 to the RX0 pin. Audio out connected to the earphones jack. 
Do I miss something here?
I enabled the ESP32 AudioKit and ES8388 in the code.

marcel-licence, 2021-11-12:
You can try to active the NoteOn in the setup function first to ensure the codec / audio is working.

KeenVox, 2021-11-12:
How do I activate the function? changing 0 to 1? If so the codec doesn't seem to work...

marcel-licence, 2021-11-12:
Yes the following line will generate a NoteOn:

```
#if 1 /* activate this line to get a tone on startup to test the DAC */
    Synth_NoteOn(0, 64, 1.0f);
#endif
```

You can check your debug output:
`Connect to ES8388 codec... ` is expected. If a "failed" follows the I2C setting might be incorrect.

If that step was successful we can check the I2S configuration. There might be a second possibility (there are two different versions containing the ES8388 codec)

marcel-licence, 2021-11-12:
In some cases
`#define I2S_BCLK    27` should be replced by `#define I2S_BCLK    5` in board_audio_kit_es8388.h
Please ensure if you try this that IO5 is left open on the pin header.

KeenVox, 2021-11-12:
This is what I get:
Connect to ES8388 codec... 0x00: 0x00
16:39:38.326 -> 0x01: 0x00
16:39:38.326 -> Failed!

#define I2S_BCLK  is on 25 in your code. WCLK on 27, which one should I change?

marcel-licence, 2021-11-12:
The codec has not been found. I would recommend to use the I2C scanner first. Do you have the black or the copper coloured antenna on your audio kit? In case of copper the AC101 should be assembled.

KeenVox, 2021-11-12:
I've got the black antenna. I tried compiling with AC101, here I get this:
Connect to AC101 codec... Failed!
16:50:12.506 -> Failed!

marcel-licence, 2021-11-12:
Okay the black antenna is an indicator for the ES8388. I would suggest to use the I2C scanner first. Ensure that nothing additional is connected to the I2C lines. 

There are different combinations of SDA and SCL you can check

KeenVox, 2021-11-12:
Okay, how do I do that?

marcel-licence, 2021-11-12:
In esp32_basic_synth.ino is a function called `void  ScanI2C(void)`. You can call this one.
The call `Wire.begin(21, 22)` provides 21, 22 as PINs for SDA and SCL

You can try different variations until the scanner is able to find a device. In that case you will get a debug message:
`I2C device found at address 0x..`

You can add `ScanI2C` in the setup function after the Serial.begin(..) call

KeenVox, 2021-11-12:
I'm not really sure what to do exactly. Do I copy the code after void ScanI2C(void) after Serial.begin? Or do I type ScanI2C before that?

KeenVox, 2021-11-12:
Ah got it. What Pins should I check?

marcel-licence, 2021-11-12:
`Wire.begin(21, 22)` and `Wire.begin(18, 23)`

KeenVox, 2021-11-12:
No devices found for both

marcel-licence, 2021-11-12:
Hm sounds like there is hopefully a datasheet available for the ESP32-A1S showing where the codec could be connected. Or that would be the second time that it is a faulty device.

I've found another configuration:
`Wire.begin(33, 32)`

marcel-licence, 2021-11-12:
And another one
`Wire.begin(27, 28)`

Hm very confusing

KeenVox, 2021-11-12:
> Hm sounds like there is hopefully a datasheet available for the ESP32-A1S showing where the codec could be connected. Or that would be the second time that it is a faulty device.
> 
> I've found another configuration: `Wire.begin(33, 32)`

That one said device found at 0x10, does that help?

marcel-licence, 2021-11-12:
Yes. Now you can update the defines for the ES8388 I2C.
In the file "board/board_audio_kit_es8388.h" you will find
```
#define I2C_SDA 18
#define I2C_SCL 23
```
that should be replaced by

```
#define I2C_SDA 33
#define I2C_SCL 32
```

KeenVox, 2021-11-12:
Didn't work still:
Connect to ES8388 codec... 0x00: 0x00
18:17:20.505 -> 0x01: 0x00
18:17:20.505 -> Failed!

Anything left, we could try?

marcel-licence, 2021-11-12:
Yes I picket the wrong define.
In es8388.ino you will find:
```
#define ES8388_PIN_SDA  18
#define ES8388_PIN_SCL  23
```
and should be replaced by

```
#define ES8388_PIN_SDA  33
#define ES8388_PIN_SCL  32
```

KeenVox, 2021-11-12:
yes that worked!
Connect to ES8388 codec... 0x00: 0x06
18:33:44.015 -> 0x01: 0x5c
18:33:44.015 -> AdcCh1!
18:33:44.015 -> MixChAMPL!
18:33:44.015 -> ES8388 setup finished

But soundwise I only get a crack sound form the earphone jack on startup.

marcel-licence, 2021-11-12:
The NoteOn is still active?
In that case the I2S pinning could be also incorrect.

You can try to replace in es8388.ino:

```
#define ES8388_PIN_MCLK 0
#define ES8388_PIN_SCLK 5
#define ES8388_PIN_LRCK 25
#define ES8388_PIN_DIN  26
#define ES8388_PIN_DOUT 35
```
by
```

#define ES8388_PIN_MCLK 0
#define ES8388_PIN_SCLK 27
#define ES8388_PIN_LRCK 25
#define ES8388_PIN_DIN  26
#define ES8388_PIN_DOUT 35
```

KeenVox, 2021-11-12:
That worked :) What a hassle. 
Midi still doesn't work sadly. Do I have to change pins too?

marcel-licence, 2021-11-12:
Haha ... that is the reason I've started creating my first PCB xD
How did you connect MIDI? There is also a debug option to generate output to get a feedback of received messages:
`MIDI_DUMP_SERIAL2_TO_SERIAL` in midi_interface.ino

KeenVox, 2021-11-12:
I use a DIN connection with an 6N138. I take 3.3v and GND from the Audiokit pins and conected the 6n138 out to the RX0 pin, but it won't work.

marcel-licence, 2021-11-12:
RX0 is not MIDI. It is also connected to the serial usb converter.
`#define MIDI_RX_PIN 19` defined in board_audio_kit_es8388.h

KeenVox, 2021-11-12:
Ahhh, damn!
Ok, now I connected everything to pin19, the LED is blinking when I use my keyboard. But theres no audio output, what could be the problem now? I'm becoming insande... :-D

marcel-licence, 2021-11-12:
Sounds good. Looks like there is only a little piece missing. Did you get any debug output?
Ah maybe the filter (cutoff) has been reset to 0

KeenVox, 2021-11-12:
I get no errors in the terminal. I activated the midi dump, seems normal shows activity when I press the keys. But still no sound to the earphone output. 
When I reactivate the test sound it works fine.

KeenVox, 2021-11-12:
on the other hand it seems I always get the same output on every key
ff00>c0 00 00
19:41:39.804 -> ff00>c0 00 00
19:41:39.804 -> ff00>c0 00 00
19:41:39.804 -> ff00>c0 00 00
19:41:39.850 -> ff00>c0 00 00
19:41:39.850 -> ff00>c0 00 00

marcel-licence, 2021-11-12:
Okay looks like a baudrate or an issue with your circuit. The bits are not detected

KeenVox, 2021-11-12:
hm I used a circuit for an Arduino, got a good cirucit I should try?

marcel-licence, 2021-11-12:
Did you connect a pull up to IO19? FF and 00 indicates that the signal is changing slowly. You should also ensure that the signal goes never above 3.3V on the IO19

KeenVox, 2021-11-12:
Okay, now I everthing broke, I can't get any midi signals to the audi kit anymore. The test soudn works though...

KeenVox, 2021-11-12:
okay back on track midi seem to work:
2600>80 26 00
21:09:00.558 -> 2000>80 20 00
21:09:01.073 -> 0026>80 00 26
21:09:01.073 -> 0000>80 00 00
21:09:01.167 -> 2600>80 26 00
21:09:01.994 -> 0030>80 00 30
21:09:01.994 -> 3c00>80 3c 00
21:09:02.135 -> 3000>80 30 00

But still no audio output :-(

marcel-licence, 2021-11-12:
Hm there is no NoteOn event.
For example pressing a key:
9c nn vv (c: channel, nn: note, vv: velocity)
Note off:
8c nn vv (c: channel, nn: note, vv: may be zero)
Some devices also sending a note on with velocity = 0 in case of a note off.

KeenVox, 2021-11-12:
hm that's strange. It's an Aturia Keystep and it works with all my synths, pc, and arduino and teensy synths. I even build you synth with the Devkit and an DAC and it did work. 
Any ideas left? I have got the feeling we a just in front of the finish line

marcel-licence, 2021-11-12:
Yeah. If your other arduino or teensy synth also works with 3.3V you can connect IO19 to the rx of another synth. That will confirm that the signal will be okay. Is your optocoupler supplied by 3.3V of the audio kit?

KeenVox, 2021-11-12:
yes, I get 3,45 on the optocoupler. I've even used a H1L11 instead of a 6N138 because it's rated speciffically for 3,3V.

marcel-licence, 2021-11-12:
I've also used the H1L11. On a breadboard I had sometimes issues. After making a small soldered circuit the problems where gone. But I am not sure if it is a hardware problem

KeenVox, 2021-11-12:
I don't know how, at least it seems to send midi data, I tried another midi controller, exactly the same thing. Shows messages but no sound. :-(

marcel-licence, 2021-11-12:
Hm I would suggest to try something like this:

```
Serial2.begin(MIDI_SERIAL2_BAUDRATE, SERIAL_8N1, MIDI_RX_PIN);
while(true)
{
  if (Serial2.available())
  {
   Serial.printf(" %02x", Serial2.read());
  }
}
```

To record a little dump to see what data was received.

KeenVox, 2021-11-12:
Where should I put this?

marcel-licence, 2021-11-12:
In the loop function as first statement. In that case you need only the while ... Serial2.begin has been already called at that state by the MidiInit.

KeenVox, 2021-11-12:
Hm I don't really understand where to put it, I tried some positions but on the other hand, I don't really get what I should see...sry

KeenVox, 2021-11-12:
Initialize Midi Module
22:27:27.259 -> Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
22:27:27.259 -> Core 1 register dump:
22:27:27.259 -> PC      : 0x40081deb  PS      : 0x00050034  A0      : 0x4008751c  A1      : 0x3ffbe7a0  
22:27:27.306 -> A2      : 0x3ffbefa4  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x4008d6f4  
22:27:27.306 -> A6      : 0x00000001  A7      : 0x00000000  A8      : 0x3ff40000  A9      : 0x00000008  
22:27:27.306 -> A10     : 0x3ffbe7d8  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3ffb1e30  
22:27:27.306 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000015  EXCCAUSE: 0x00000006  
22:27:27.306 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
22:27:27.306 -> Core 1 was running in ISR context:
22:27:27.352 -> EPC1    : 0x400d4af1  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40081deb
22:27:27.352 -> 
22:27:27.352 -> Backtrace: 0x40081deb:0x3ffbe7a0 0x40087519:0x3ffbe7d0 0x400d4aee:0x3ffb1e90 0x400d4b51:0x3ffb1eb0 0x400d4c5c:0x3ffb1ed0 0x400d3e25:0x3ffb1ef0 0x400d3e4d:0x3ffb1f10 0x400d253a:0x3ffb1f50 0x400d2b4f:0x3ffb1f80 0x400d4573:0x3ffb1fb0 0x4008afb1:0x3ffb1fd0
22:27:27.352 -> 
22:27:27.352 -> Core 0 register dump:
22:27:27.352 -> PC      : 0x400f43b2  PS      : 0x00060134  A0      : 0x800d7116  A1      : 0x3ffbbff0  
22:27:27.352 -> A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001  
22:27:27.399 -> A6      : 0x00060120  A7      : 0x00000000  A8      : 0x800d6c26  A9      : 0x3ffbbfc0  
22:27:27.399 -> A10     : 0x00000000  A11     : 0x40087df4  A12     : 0x00060120  A13     : 0x3ffbb690  
22:27:27.399 -> A14     : 0x00000000  A15     : 0x3ffbbce0  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
22:27:27.399 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
22:27:27.399 -> 
22:27:27.399 -> Backtrace: 0x400f43b2:0x3ffbbff0 0x400d7113:0x3ffbc010 0x4008c752:0x3ffbc030 0x4008afb1:0x3ffbc050
22:27:27.399 -> 
22:27:27.399 -> Rebooting...
22:27:27.399 -> ets Jun  8 2016 00:22:57
22:27:27.399 -> 
22:27:27.399 -> rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
22:27:27.446 -> configsip: 0, SPIWP:0xee
22:27:27.446 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:27:27.446 -> mode:DIO, clock div:1
22:27:27.446 -> load:0x3fff0018,len:4
22:27:27.446 -> load:0x3fff001c,len:1216
22:27:27.446 -> ho 0 tail 12 room 4
22:27:27.446 -> load:0x40078000,len:9720
22:27:27.446 -> ho 0 tail 12 room 4
22:27:27.446 -> load:0x40080400,len:6352
22:27:27.446 -> entry 0x400806b8

Is this what we need?

marcel-licence, 2021-11-12:
Oh I see a delay was missing in the code. I didn't test it because I am currently working on another project.

```
while(true)
{
  if (Serial2.available())
  {
   Serial.printf(" %02x", Serial2.read());
  }
delay(20);
}
```
This should fix the watchdog triggered reboot.

KeenVox, 2021-11-12:
Initialize Midi Module
22:50:07.310 -> Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
22:50:07.357 -> Core 1 register dump:
22:50:07.357 -> PC      : 0x40081deb  PS      : 0x00050034  A0      : 0x4008751c  A1      : 0x3ffbe7a0  
22:50:07.357 -> A2      : 0x3ffbefcc  A3      : 0x00000002  A4      : 0x00000001  A5      : 0x4008d6f4  
22:50:07.357 -> A6      : 0x00000001  A7      : 0x00000000  A8      : 0x3ff6e000  A9      : 0x00000008  
22:50:07.357 -> A10     : 0x00000180  A11     : 0x00000180  A12     : 0x00000000  A13     : 0x3ffb1e30  
22:50:07.357 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000015  EXCCAUSE: 0x00000006  
22:50:07.357 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
22:50:07.404 -> Core 1 was running in ISR context:
22:50:07.404 -> EPC1    : 0x400d4af1  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40081deb
22:50:07.404 -> 
22:50:07.404 -> Backtrace: 0x40081deb:0x3ffbe7a0 0x40087519:0x3ffbe7d0 0x400d4aee:0x3ffb1e90 0x400d4b55:0x3ffb1eb0 0x400d4c60:0x3ffb1ed0 0x400d3e29:0x3ffb1ef0 0x400d3e51:0x3ffb1f10 0x400d253c:0x3ffb1f50 0x400d2b53:0x3ffb1f80 0x400d4577:0x3ffb1fb0 0x4008afb1:0x3ffb1fd0
22:50:07.404 -> 
22:50:07.404 -> Core 0 register dump:
22:50:07.404 -> PC      : 0x400f43b6  PS      : 0x00060134  A0      : 0x800d711a  A1      : 0x3ffbbff0  
22:50:07.451 -> A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001  
22:50:07.451 -> A6      : 0x00060120  A7      : 0x00000000  A8      : 0x800d6c2a  A9      : 0x3ffbbfc0  
22:50:07.451 -> A10     : 0x00000000  A11     : 0x40087df4  A12     : 0x00060120  A13     : 0x3ffbb690  
22:50:07.451 -> A14     : 0x00000000  A15     : 0x3ffbbce0  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
22:50:07.451 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
22:50:07.451 -> 
22:50:07.451 -> Backtrace: 0x400f43b6:0x3ffbbff0 0x400d7117:0x3ffbc010 0x4008c752:0x3ffbc030 0x4008afb1:0x3ffbc050
22:50:07.498 -> 
22:50:07.498 -> Rebooting...
22:50:07.498 -> ets Jun  8 2016 00:22:57
22:50:07.498 -> 
22:50:07.498 -> rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
22:50:07.498 -> configsip: 0, SPIWP:0xee
22:50:07.498 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:50:07.498 -> mode:DIO, clock div:1
22:50:07.498 -> load:0x3fff0018,len:4
22:50:07.498 -> load:0x3fff001c,len:1216
22:50:07.498 -> ho 0 tail 12 room 4
22:50:07.498 -> load:0x40078000,len:9720
22:50:07.498 -> ho 0 tail 12 room 4
22:50:07.498 -> load:0x40080400,len:6352
22:50:07.498 -> entry 0x400806b8

KeenVox, 2021-11-12:
seems to be the same or I copied it in the wrong place. If I put it after void loop () nothing seems to happen. It just stops with ff

marcel-licence, 2021-11-12:
Hm should look like this:

Initialized
Initialize I2S Module
Initialize Midi Module
Turn off Wifi/Bluetooth
ESP.getFreeHeap() 140740
ESP.getMinFreeHeap() 140720
ESP.getHeapSize() 286572
ESP.getMaxAllocHeap() 69376
Firmware started successfully
 fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe fe 90 32 3c fe 80 32 15 fe fe 90 34 34 fe 80 34 0f 90 35 49 fe 90 37 24 80 35 18 fe 80 37 3e 90 39 4e fe 90 37 35 80 39 1c 90 35 52 80 37 42 90 34 64 fe 80 35 43 34 56 fe fe fe fe fe fe fe fe fe fe

marcel-licence, 2021-11-12:
Generated with this code:
```
void loop()
{
    while (true) /* this is new */
    {
        if (Serial2.available())
        {
            Serial.printf(" %02x", Serial2.read());
        }
        delay(20);
    }

...
```

KeenVox, 2021-11-12:
If I try that I get 'Midi_SendShortMessage' was not declared in this scope
while compiling.

KeenVox, 2021-11-12:
ah my bad, now I get this:
23:03:18.502 -> Initialize Synth Module
23:03:18.502 -> Initialize I2S Module
23:03:18.502 -> Connect to ES8388 codec... 0x00: 0x05
23:03:18.594 -> 0x01: 0x40
23:03:18.640 -> AdcCh1!
23:03:18.640 -> MixChAMPL!
23:03:18.640 -> ES8388 setup finished!
23:03:18.640 -> Reg 0x00 = 0x05
23:03:18.640 -> Reg 0x01 = 0x40
23:03:18.640 -> Reg 0x02 = 0x00
23:03:18.640 -> Reg 0x03 = 0x00
23:03:18.640 -> Reg 0x04 = 0x3c
23:03:18.640 -> Reg 0x05 = 0x00
23:03:18.640 -> Reg 0x06 = 0x00
23:03:18.640 -> Reg 0x07 = 0x7c
23:03:18.640 -> Reg 0x08 = 0x00
23:03:18.640 -> Reg 0x09 = 0x88
23:03:18.640 -> Reg 0x0a = 0x50
23:03:18.640 -> Reg 0x0b = 0x02
23:03:18.640 -> Reg 0x0c = 0x0c
23:03:18.640 -> Reg 0x0d = 0x02
23:03:18.640 -> Reg 0x0e = 0x30
23:03:18.640 -> Reg 0x0f = 0x20
23:03:18.640 -> Reg 0x10 = 0x00
23:03:18.640 -> Reg 0x11 = 0x00
23:03:18.640 -> Reg 0x12 = 0x16
23:03:18.640 -> Reg 0x13 = 0xb0
23:03:18.640 -> Reg 0x14 = 0x32
23:03:18.640 -> Reg 0x15 = 0x06
23:03:18.640 -> Reg 0x16 = 0x00
23:03:18.640 -> Reg 0x17 = 0x18
23:03:18.640 -> Reg 0x18 = 0x02
23:03:18.640 -> Reg 0x19 = 0x22
23:03:18.640 -> Reg 0x1a = 0x00
23:03:18.640 -> Reg 0x1b = 0x02
23:03:18.688 -> Reg 0x1c = 0x08
23:03:18.688 -> Reg 0x1d = 0x00
23:03:18.688 -> Reg 0x1e = 0x1f
23:03:18.688 -> Reg 0x1f = 0xf7
23:03:18.688 -> Reg 0x20 = 0xfd
23:03:18.688 -> Reg 0x21 = 0xff
23:03:18.688 -> Reg 0x22 = 0x1f
23:03:18.688 -> Reg 0x23 = 0xf7
23:03:18.688 -> Reg 0x24 = 0xfd
23:03:18.688 -> Reg 0x25 = 0xff
23:03:18.688 -> Reg 0x26 = 0x1b
23:03:18.688 -> Reg 0x27 = 0xb8
23:03:18.688 -> Reg 0x28 = 0x28
23:03:18.688 -> Reg 0x29 = 0x28
23:03:18.688 -> Reg 0x2a = 0xb8
23:03:18.688 -> Reg 0x2b = 0x80
23:03:18.688 -> Reg 0x2c = 0x00
23:03:18.688 -> Reg 0x2d = 0x00
23:03:18.688 -> Reg 0x2e = 0x1e
23:03:18.688 -> Reg 0x2f = 0x1e
23:03:18.688 -> Reg 0x30 = 0x1e
23:03:18.688 -> Reg 0x31 = 0x1e
23:03:18.688 -> Reg 0x32 = 0x00
23:03:18.688 -> Reg 0x33 = 0xaa
23:03:18.688 -> Reg 0x34 = 0xaa
23:03:18.688 -> Initialize Midi Module
23:03:18.688 -> Turn off Wifi/Bluetooth
23:03:18.688 -> ESP.getFreeHeap() 118904
23:03:18.688 -> ESP.getMinFreeHeap() 118884
23:03:18.688 -> ESP.getHeapSize() 267096
23:03:18.688 -> ESP.getMaxAllocHeap() 69376
23:03:18.735 -> Firmware started successfully
23:03:18.735 ->  ff

marcel-licence, 2021-11-12:
Okay the ff is your first received byte via midi. If you hit any keys you should get some output

KeenVox, 2021-11-12:
yes looks like that:
23:15:30.355 -> Firmware started successfully
23:15:30.355 ->  ff 00 20 00 00 20 00

marcel-licence, 2021-11-12:
Hm okay that is really strange. There is no note on or note off event. There are some possible problems left:
- wrong io
- wrong baud rate
- hardware circuit

You could define another PIN `MIDI_RX_PIN`

The baudrate is defined by `MIDI_SERIAL2_BAUDRATE`

KeenVox, 2021-11-12:
what baud rate should I try? and the io pin? Just on of the other pins of teh audiokit?

[11 more replies]

closed 2021-11-13 by marcel-licence