marcel-licence/esp32_basic_synth #76: Compatibility with Espressif ESP32-LyraT dev board / ES8388 codec

thomasfla, 2021-12-04:
Thanks for the amazing project, I'm glad I looked for existing ESP32 based synth before starting from scratch!

I'm wondering if someone already tried to use this code on the esp32-lyrat dev board https://www.espressif.com/en/products/devkits/esp32-lyrat dev board
It integrates a different DAC: ES8388 witch is a full coded audio chip integrating also analog inputs, headphone amplifier and analog mixers. I guess it should work since it also uses I2S. Any advice is welcome.

I will try to set it up and report here, first I need to find out where I put my optocouplers..

Thanks again !

marcel-licence, 2021-12-04:
Hi,

the ES8388 is supported. The newer version of the ESP32 Audio Kit uses also the ES8388 DAC. You can check the board folder. I've prepared different configuration. I would be happy to know which configuration would be necessary for the LyraT board. In that case I could add this to the project.

You are welcome!

thomasfla, 2021-12-04:
Thanks, I should have looked at the code first..

Ok so I set up a midi input on Uart 2 RX (GPIO16), defined `BOARD_ESP32_AUDIO_KIT_ES8388` in `config.h`
and edited `board_audio_kit_es8388.h` to set RX back to `#define MIDI_RX_PIN 16`

The usb serial output look like:

```
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

esp32_basic_synth  Copyright (C) 2021  Marcel Licence
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions; 
Initialize Synth Module
Initialize I2S Module
Connect to ES8388 codec... 0x00: 0x05
0x01: 0x40
AdcCh1!
MixChAMPL!
ES8388 setup finished!
Reg 0x00 = 0x05
Reg 0x01 = 0x40
Reg 0x02 = 0x00
...
Reg 0x32 = 0x00
Reg 0x33 = 0xaa
Reg 0x34 = 0xaa
Initialize Midi Module
Turn off Wifi/Bluetooth
ESP.getFreeHeap() 116192
ESP.getMinFreeHeap() 116172
ESP.getHeapSize() 264876
ESP.getMaxAllocHeap() 69376
Firmware started successfully
```

If I enable 
```
#if 1 /* activate this line to get a tone on startup to test the DAC */
    Synth_NoteOn(0, 64, 1.0f);
#endif
```
I hear the note on the headphone jack output, so that's good !

However, It seems that I don't have any incoming midi frame. Should I expect an output on the serial monitor when a key is pressed? 

I've checked the signal on pin 16 with a scope:
[image]
Apart from the rather slow edges, it looks like a good 3.3V Midi frame? Do you have an idea of what could be the problem?

Thanks,
Thomas.

marcel-licence, 2021-12-04:
Sounds good. You can enable `//#define MIDI_DUMP_SERIAL2_TO_SERIAL` in midi_interface.ino. In addition to that you can check that MIDI_SERIAL2_BAUDRATE is set to the correct value. Usually it should default to 31250 if not set elsewhere.

The signal looks good on the first glance. I think it may be the pin or baudrate which may cause the problem.

thomasfla, 2021-12-05:
Thanks, it finally works. 

The issue was that on wrover modules, GPIO 16 and 17 are not accessible (used internally for PSRAM)
I connected the MIDI_RX on pin 22, and it works well.

marcel-licence, 2021-12-05:
I am very happy to hear that.

I can prepare a new board file for the LyraT and push it to all projects.

Is the following configuration correct?
```
#define BLINK_LED_PIN   22 /* Green LED indicator */

#define ES8388_PIN_DOUT 35
#define ES8388_PIN_DIN  26
#define ES8388_PIN_LRCK 25
#define ES8388_PIN_SCLK 5
#define ES8388_PIN_MCLK 0

#define ES8388_PIN_SDA  18
#define ES8388_PIN_SCL  23

#define MIDI_RX_PIN 22
```

It looks like that DIN and LRCK are swapped in comparison to the audio kit.

thomasfla, 2021-12-05:
Yes, the default configuration was #define ES8388_CFG_I2S  4 in the es8388.h which correspond to:

 #elif ES8388_CFG_I2S==4
#define ES8388_PIN_DOUT 35
#define ES8388_PIN_DIN  26
#define ES8388_PIN_LRCK 25
#define ES8388_PIN_SCLK 5
#define ES8388_PIN_MCLK 0
#endif

So I had nothing to change for the I2S

BTW, unfortunately on the Lyrat, we don't have access to the unused pin, so we have to solder directly to the pin 22:

[image]

Thanks for adding the config

marcel-licence, 2021-12-05:
Perfect. I will prepare the config file soon.
Hm that is bad that there is no header. It is confusing that for example a header is available for I2S (which is connected to the codec) but no free IO. I've checked the schematic. Very confusing..

Thank you very much for your investigation =)

marcel-licence, 2021-12-09:
I pushed the config file to GitHub today

closed 2021-12-09 by marcel-licence