marcel-licence/esp32_basic_synth #81: Init of Es8388 failed on audiokit v2.2 board

udok7of9, 2022-02-23:
i can compile the code and upload it.  But then i got an  error message while the es8388 is initialising.

esp32_basic_synth  Copyright (C) 2021  Marcel Licence
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions; 
Initialize Synth Module
saw ready
rev: 15329, 15329
Reverb is ready!
Initialize Audio Interface
Connect to ES8388 codec... 0x00: 0x00
0x01: 0x00
Failed!
0x00: 0x00
0x01: 0x00
Failed!

any hint ?
i tryed to get the board jumper setting from your videos but quality was too poor.

Thanks for the help.  you're projects are awesome!

marcel-licence, 2022-02-23:
There are different configurations possible:

In https://github.com/marcel-licence/ML_SynthTools/blob/main/src/boards/board_audio_kit_es8388.h you will find
#define ES8388_CFG_I2C 1
You can change the number. You can change the number and try until the ES8388 connection does not return a failure.

udok7of9, 2022-02-25:
Hello Marcel,
thanks. that helps. no i got connection on pins 32 & 33.

marcel-licence, 2022-02-25:
Perfect. I guess you mean "now", right?

closed 2022-02-25 by marcel-licence

udok7of9, 2022-02-25:
hi Marcel,
yes "Now".   i now got it running.  I need to adapt it to my controller but that is just a question of effort.  thank you for this wonderful peace of software.   BTW.  i would like to implement the scope but the code is still missing. can't wait for the update.