marcel-licence/esp32_basic_synth #116: Midi keyboard connection wiring

ramainen, 2024-04-19:
Hi! How you connect your midi keyboard to board? I have one esp 32 audio kit board, but I did not understand, which board pin must be connected with midi pin.

also, in this video https://youtu.be/_iwpOnLWirU?t=574 I see something orange, midi-like. Is this mean I can connect midi to audio-in jack?...

Thank you

ramainen, 2024-04-19:
6N138 maybe?...

marcel-licence, 2024-04-19:
Hi,
yes, a little circuit is required to connect your MIDI device. The most recommended part is this one: https://github.com/marcel-licence/ML_SynthTools/blob/platformIO/extras/MIDI_IN_H11L1.png
You will find little schematics for the 6N138 too.
The orange cable is a DIN 5pin to 3.5mm jack adapter for easier connection with the circuit. You can use a DIN 5pin socket too or connect the two cables to your circuit directly.
Hope this helps

ramainen, 2024-04-25:
My midi cable has 5 and 4 pins connected to cable. 3, 2 and 1 connected to nothing. But in chematics https://github.com/marcel-licence/ML_SynthTools/blob/platformIO/extras/MIDI_IN_H11L1.png wires are connected to 3 and 4
[image]
[image]

And in another sources also using 4 and 5: https://electronics.stackexchange.com/questions/400080/how-do-i-use-h11l1-correctly-to-receive-midi and https://forum.arduino.cc/t/midi-3-3v-circuit/698917

Maybe mistake there?

ramainen, 2024-04-25:
I see, must be 4 and 5.

ramainen, 2024-04-26:
And I have another question, about correct pin to connect MIDI to board.

I have ESP32 Audio Kit board 2.2 (with es8388).

In this schema https://github.com/marcel-licence/ML_SynthTools/blob/platformIO/extras/MIDI_IN_H11L1.png written "Connect to RX pin". But if Ill connect midi to RXR0, usb debugging will be not work? Can I change MIDI IN pin? 

I did not found PIN constant here https://github.com/marcel-licence/ML_SynthTools/blob/platformIO/src/boards/board_audio_kit_es8388.h

closed 2024-04-29 by ramainen

marcel-licence, 2024-05-15:
Yes, thank you for your feedback. It was a mistake. I updated the schematic today. I didn't recognized the wrong connection.
Please let me know if this issue is solved now

ramainen, 2024-05-15:
Totally solved, esp32_basic_synth works well, connected 4 and 5

marcel-licence, 2024-05-15:
Perfect! =)