marcel-licence/esp32_basic_synth #118: Fehler beim Kompilieren: undefined reference to `Status_ValueChangedFloat(char const*, char const*, float)'

renatoni, 2025-02-07:
Hallo Marcel,
das Kompilieren des Projektes bricht mit folgenden Fehler ab:
the compilation of the project aborts with the following error:

C:/Users/renato.nitsch/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:\Users\renato.nitsch\AppData\Local\arduino\sketches\B26938C3E60C34EE99926F57F7DFFCD4\sketch\ml_reverb.cpp.o:(.literal._Z15Reverb_SetLevelhf+0x8): undefined reference to `Status_ValueChangedFloat(char const*, char const*, float)'
C:/Users/renato.nitsch/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:\Users\renato.nitsch\AppData\Local\arduino\sketches\B26938C3E60C34EE99926F57F7DFFCD4\sketch\ml_reverb.cpp.o: in function `Reverb_SetLevel(unsigned char, float)':
C:\Users\renato.nitsch\Documents\Arduino\esp32_basic_synth_ml/ml_reverb.cpp:271:(.text._Z15Reverb_SetLevelhf+0x10): undefined reference to `Status_ValueChangedFloat(char const*, char const*, float)'
collect2.exe: error: ld returned 1 exit status

Habe leider nicht genug Erfahrung um das Problem selbst beheben zu können.
Ich danke dir für Deine Hilfe.

Unfortunately I don't have enough experience to fix the problem myself.
Thank you for your help.

Grüße
Renato

marcel-licence, 2025-02-11:
Hallo Renato (@renatoni ),

für den Soundtest kann ich nur empfehlen, die folgende Zeile zu aktivieren:
`#define OUTPUT_SAW_TEST`
Damit wird ein Sägezahn erzeugt.

Vielleicht stimmt etwas mit der Verkabelung nicht. Beim Start kann man im Serial Monitor die Pin Belegung auch angezeigt bekommen.

Viele Grüße
Marcel

renatoni, 2025-02-13:
Hallo Marcel, danke für Deine Nachricht.
Habe den Soundtest (#define OUTPUT_SAW_TEST)  ausprobiert. Es wird, nach dem Durchlauf der Setupprozedur, ein Dauerton ausgegeben.
Das scheint ja zumindest bei meinem Aufbau zu funktionieren.

In der im Projekt angegebenen Schaltung hat die LED am RX2 keinen Vorwiderstand. Ist das so gewollt oder nur stark vereinfacht dargestellt? Ich habe da mal einen 1KOhm Widerstand  zwischen 3,3V und Anode der LED (High efficient LED) eingefügt. Flakkerte dann bei MIDI-Signal immer noch ausreichend.

Wird das eingehende MIDI-Signal im "Normalfall" programmtechnisch auf den TX2 durchgeschleift, also dort umgehend wieder ausgegeben?

marcel-licence, 2025-03-09:
Hallo Renato,
das MIDI signal sollte eigentlich nicht durchgeschliffen werden. Man könnte es machen, müsste aber dazu den Code implementieren. Um ganz sicher zu gehen, müsste ich sonst auch noch einmal schauen.
MIDI Eingänge hatte ich hier https://github.com/marcel-licence/ML_SynthTools/blob/main/extras/midi_input.md dokumentiert.
Es stimmt, der Widerstand fehlt in dem Schaltbild. Ich denke ich werde es entfernen und zentral bei den ML_SynthTools dokumentieren.
Viele Grüße

renatoni, 2026-02-20:
Hallo Marcel, ich habe das leider nicht hin bekommen. Schaltungstechnisch war soweit alles in Ordnung es konnte aber über MIDI kein ton entlockt werden. Hatte das Projekt aus Zeitgründen (Familie) erstmal auf Eis gelegt. Werde das aber wieder aufnehmen sobald es wieder genug Zeit für diese "Basteleien" gibt. 

Viele Grüße und danke für Deine Bemühungen.
Renato

marcel-licence, 2026-02-20:
Hallo Renato,
vielen Dank für die Rückmeldung. Wenn es wieder los geht werde ich gerne unterstützen. Ich werde das Projekt auch noch einmal überprüfen und das Ticket hier erst einmal schliessen.
Viele Grüße
Marcel

renatoni, 2026-02-22:
Hallo Marcel,
alles soweit klar, ich melde mich sobald ich wieder an dem Projekt arbeite.

Grüße
Renato